robotpy/robotpy-wpilib-utilities #22: magicbot: add feedback tunable

virtuald, 2016-05-15:
Example:

```
class Component:
    # this would automatically set an NT value at /components/component/angle
    @feedback
    def get_angle(self):
        return ... 
```

There's a few things I would like to solve:

* Need to provide sensor feedback in disabled mode -- would like to avoid adding extra user code to make that happen
* Would like to cache variables read at the beginning of each control loop, to avoid potential race conditions

Twinters007, 2016-12-31:
Would the syntax need to be `get_xxx` and NT value be /components/component/xxx?

virtuald, 2017-01-02:
I think the only thing that would need to be required is the `@feedback` decorator. I don't think I would want *every* get_ function to be added, otherwise you could accidentally trigger it.

Twinters007, 2017-01-02:
I mean in order to get a key to put in nt, would you look for whatever comes after get_?

virtuald, 2017-01-02:
Yeah, I think you could autodetect that. The feedback decorator would probably take an optional keyword parameter so that one could override it if they wanted.

Twinters007, 2017-09-05:
Close?

auscompgeek, 2017-12-08:
Fixed in #66. Closing.

closed 2017-12-08 by auscompgeek