robotpy/robotpy-wpilib-utilities #26: Higher and lower level components is confusing

calebeby, 2016-09-06:
Hi @virtuald and @PeterJohnson,
I think it's confusing that there are higher and lower level components. I like the way that @frc1418 had `components` and `automations`. I think updating this in the docs would make it easier for beginners to understand the difference, and it would be good for them to live in separate folders.

virtuald, 2016-09-06:
Is it the wording that's confusing or the concept?

calebeby, 2016-09-06:
The wording

virtuald, 2016-09-06:
Ok, that's fair then.

I'm not quite sure how I feel about using the word 'automations'. I chose 'components' and 'controllers' for my rewrite of [2423's code](https://github.com/frc2423/2016/tree/master/robot).

It's probably a good idea to add a note that it's useful to separate them into different folders. Though, worth noting that the [magicbot sample](https://github.com/robotpy/robotpy-wpilib-utilities/tree/master/samples/magicbot_simple) only has a components directory -- does that mean we should add a more complex magicbot example?

I'm working on other things, but in the meantime feel free to edit it yourself and submit a PR to clarify it. :)

calebeby, 2016-09-07:
Regarding the name, personally, I prefer automations, because you are connecting more than one component together.

Regarding examples, reading [1418's code](https://github.com/frc1418/2016-robot/tree/master/robot) has been more helpful for me than the samples because it has real parts, like `intake`, `winch`, and `drive`, instead of `component1` and `component2`.
I definitely think it would be helpful to create a more realistic example that is simpler than 1418's code.

I am currently programming a robot for [BunnyBots](http://team1540.org/bunnybots/), and once that season is over (middle of December) I would be willing to add it as an example.

@ErikBoesen what do you think about the naming? Do you prefer `automations`, `controllers`, or something else?

ErikBoesen, 2016-09-07:
I don't have a strong opinion on this, but if I had to choose I'd go with "automations," even though it's rather confusing when combined with autonomous.