robotpy/robotpy-wpilib-utilities #33: onException should have a context manager wrapper version

auscompgeek, 2016-12-28:
This is a call for feedback on an idea I've been pondering for a while.

Currently, the recommended thing to do to handle uncaught exceptions is to wrap blocks of code in a try-except:
```python
def teleopPeriodic(self):  # or whatever method
    try:
        doSomething()
    except:
        self.onException()
```

I feel this is two lines too much boilerplate. Hence, I propose the addition of a context generator, which will consume any exceptions raised, as a new way of handling uncaught exceptions.

However, I'm slightly stuck on what this should look like. There are two alternatives I've been thinking about:

```python
# The MagicBot class is the context manager.
def teleopPeriodic(self):
    with self:
        doSomething()
```

```python
# The MagicBot class has a consumeExceptions() method which returns a context manager.
def teleopPeriodic(self):
    with self.consumeExceptions():
        doSomething()
```

Both would be trivial to implement, but I'm leaning towards the second option because it's clearer. What does everyone think?

james-ward, 2016-12-28:
I agree with your predilection! Option two for me.

virtuald, 2016-12-29:
Option two for me. And add docs to the magicbot page in robotpy-docs.