robotpy/robotpy-wpilib-utilities #39: Access the magicbot component constructor

CarterFendley, 2017-01-22:
**Use case:** I have two objects that control swerve drive. The SwerveModule class controls individual modules. The SwerveDrive class controls the overall driving of the four SwerveModules. It would be useful to have access to the constructor of SwerveModule because I need to pass it around 4 parameters per class (motor1, motor2, encoder, inverted). If I had to create object for all of these in `robot.py` I would end up with about 16 variables that I never use again.

virtuald, 2017-02-11:
I have an alternative idea for this. What if instead we used a prefix scheme to inject things? Something like...

```
class Component:
    motor = wpilib.Talon

class MyRobot:
    left_component = Component
    right_component = Component
    
    def create_objects(self):
        left_component_motor = wpilib.Talon
        right_component_motor = wpilib.Talon

```

So when it was looking for variables to inject, it looks first for the actual name, and if it doesn't find it then it looks for the variable name prefixed by the component name? This would solve your issue, yeah?

Keep in mind that you could inject anything you wanted, it doesn't have to be a motor -- it could be a tuple too I think.

virtuald, 2017-02-11:
The thing I'm undecided on is whether it should look for the prefixed version of the variable first, or whether it should look for the variable name first. For backwards compatibility, probably variable name first..

CarterFendley, 2017-02-13:
So my main issue is that there is a lot of things in my constructor. To use magic bot components I would have to create two motor object and one encoder object for each Swerve Module. Then I would have to call multiple functions to turn off and on flags and also set the zero value for the encoder. For each module that I initialized it would be around 8 lines of code. I would end up with around 32 lines of just creating swerve module things.

CarterFendley, 2017-02-13:
I would be happy if some how we had access to the arguments passed to the `setup()` method.

virtuald, 2017-02-13:
You could do:

```
class MyRobot:
    module_args = (1,2,3,4)

class Component:
    module_args = tuple
```

virtuald, 2017-02-16:
Turns out I had already implemented the prefix thing. You should use a tuple to pass your args to the underlying component. I'll write some docs.

closed 2017-02-16 by virtuald

virtuald, 2017-02-16:
Docs are up here: http://robotpy.readthedocs.io/en/stable/frameworks/magicbot.html#variable-injection