robotpy/robotpy-wpilib-utilities #50: NavX getRate() not working

czhao39, 2017-03-04:
It seems that `getRate()` stopped working due to an update within the past two weeks. Now when the NavX is not rotating, `getRate()` returns 360 degrees per second. However, `getAngle()` still seems to work fine. Unfortunately, I did not have enough time today to play with the NavX to see if the offset of 360 degrees per second was consistent.

auscompgeek, 2017-03-04:
The NavX driver was updated in 1f4fc57 for this year. Having a look at [continuousangletracker.py](/robotpy/robotpy-wpilib-utilities/blame/1f4fc57/robotpy_ext/common_drivers/navx/continuousangletracker.py#L104) and the upstream [ContinuousAngleTracker.java](/kauailabs/navxmxp/blob/c5e3a8a9b642/roborio/java/navx_frc/src/com/kauailabs/navx/frc/ContinuousAngleTracker.java#L128), it looks like someone somehow forgot to include a condition.

czhao39, 2017-03-04:
Oh yeah, that would do it. Thanks!

virtuald, 2017-03-04:
Sorry about that, I'll push a new release momentarily.

PR #51 (2017-03-04): NavX: Fix getRate() returning 360°/s when stationary
Somehow, someone neglected to add in a condition, breaking ContinuousAngleTracker.getRate() in the process.

Fixes: #50
commit 755b9e7b (2017-03-04): NavX: Fix getRate() returning 360° when stationary

Fixes: #50
closed 2017-03-04 by virtuald