robotpy/robotpy-wpilib-utilities #82: Update navX library for 2018

auscompgeek, 2018-01-05:
There's been a number of changes in WPILib that have necessitated changes to the navX library. kauailabs/navxmxp@73aa4ea84010 in particular sticks out to me.

@virtuald I was taking a brief look at this earlier, and I noticed the commit referenced in the git-source-track line in ahrs.py is from 2016. Judging by the code, I'm guessing that wasn't intended?

virtuald, 2018-01-05:
I may have forgotten to update the git source-track line last year -- though, if I recall correctly, there were not a ton of changes.

auscompgeek, 2018-01-08:
The weird part is that the line _was_ updated. It's just a commit from a significant amount of time before the line was updated.

virtuald, 2018-01-08:
It's possible that I updated it and neglected to update git-source-track. Sounds like the sort of thing I might do.

closed 2018-02-17 by auscompgeek