robotpy/robotpy-wpilib-utilities #86: magicbot should subclass TimedRobot

auscompgeek, 2018-01-07:
Let's avoid reinventing the wheel.

twinters007, 2018-01-07:
should we make this happen in 2018? Right now, because it subclasses sample, DeprecationWarnings are getting thrown in tests

virtuald, 2018-01-07:
This seems like a good idea.

auscompgeek, 2018-01-07:
I think we should rewrite the auto framework to instead run single iterations at a time and let the robot class deal with timing. Maybe keep around a version that keeps using PreciseDelay if for some reason a team wants to keep using IterativeRobot but prefers a stable timing for autonomous.

Also on the SampleRobot front, maybe we should un-deprecate SampleRobot. WPILib's intent there was to make a warning show up in IDEs and at compile time. Showing some other warning, preferably one that can be programmatically ignored if teams so wish, could happen in simulation too. I would prefer such a warning to not show up in DS logs though.

auscompgeek, 2019-09-29:
Upstream have removed SampleRobot altogether (the replacement being to subclass RobotBase directly), so this should happen ASAP. Should be relatively straightforward.

PR #148 (2020-01-13): magicbot: Remove SampleRobot usage
:shipit:

Fixes: #86
closed 2020-01-13 by virtuald