robotpy/robotpy-wpilib-utilities #88: MagicBot StateMachine breaks when its last state is timed and has a default state

amorygalili, 2018-01-15:
In the example below I would expect after b is done the state machine would transition to either the **default** or **a** state. Instead the console shows the state machine switching back and forth between the **default** and **a** state (it never transitions from **a** to **b**). If you run this code in the simulator with teleop enabled the console shows this:

> 19:36:23:486 WARNING : robot               : Default MagicRobot.disabledPeriodic() method... Overload me!
19:36:23:642 INFO    : nt                  : CONNECTED 127.0.0.1 port 51670 (pynetworktables 2017.0.8)
19:36:23:685 WARNING : pyfrc.sim.ui        : pygame not detected, real joystick support not loaded
a
b
default
a
default
a
default
a
default
a
default

Calling the StateMachine done method seems to fix this issue, maybe it should be called automatically at the end of the final state?

Here's the code:

```python
#!/usr/bin/env python3

import wpilib
from magicbot import StateMachine, state, timed_state, MagicRobot
from magicbot.state_machine import default_state

class SM(StateMachine):

    @default_state
    def d(self):
        print('default')

    @state(first=True)
    def a(self):
        print('a')
        self.next_state('b')

    @timed_state(duration=1)
    def b(self):
        print('b')

class MyRobot(MagicRobot):

    sm = SM

    def createObjects(self):
        self.joystick = wpilib.Joystick(0)

    def teleopPeriodic(self):
        print()
        self.sm.engage()

if __name__ == "__main__":
    wpilib.run(MyRobot)

```

virtuald, 2018-01-15:
Hm, because the state machine is constantly being engaged, I would expect the output to be like so:

```
a
b
a
b
a
b
...
```

virtuald, 2018-02-17:
I wrote some tests for this, and your fix doesn't actually fix the issue. Instead of what I expected, you get the following:

```
a
b
default
a
b
default
a
b
default
...

The thing about your example is that the timed_state period (0.01) is less than the execution period (0.02), so technically the original example should be doing `a a a a a a...` because when it tries to execute the b state, it has already expired.

I think instead that we should guarantee that a timed_state will execute at least once, as this is a more reasonable behavior. I'll create some tests and push out a more complete fix.

commit e0631bd9 (2018-01-16): Fix timed states expiring immediately when run twice (#89)

Fixes: #88
commit 348752ad (2018-02-17): Magicbot: guarantee that timed_state will execute at least once

- Also ensures that default_state will not execute if the state machine is always executing
- Reverts e8c7a3c0c4536e91ec63559865962f7079e74d78
- Fixes #88
PR #107 (2018-02-17): Magicbot: guarantee that timed_state will execute at least once
- Also ensures that default_state will not execute if the state machine is always executing
- Reverts e8c7a3c0c4536e91ec63559865962f7079e74d78
- Fixes #88
closed 2018-02-17 by virtuald