robotpy/robotpy-wpilib-utilities #96: Console fills with NavX errors (NavX still works)

vanjac, 2018-02-06:
Both when the robot is disabled and enabled, the console is constantly being flooded with almost identical messages:

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/robotpy_ext/common_drivers/navx/registerio.py", line 114, in run
    self.getCurrentData()
  File "/usr/local/lib/python3.6/site-packages/robotpy_ext/common_drivers/navx/registerio.py", line 176, in getCurrentData
    curr_data = self.io_provider.read(first_address, read_count)
  File "/usr/local/lib/python3.6/site-packages/robotpy_ext/common_drivers/navx/registerio_spi.py", line 85, in read
    raise IOError("CRC error")
OSError: CRC error

Locals at innermost frame:

{ 'count': 108,
  'crc': 237,
  'data': [ 123,
            123,
... continued ...
```

The NavX works fine as far as I can tell (I've only tested `getAngle()`). But these constant errors make it difficult to debug any other problems.

vanjac, 2018-02-06:
The issue was actually in our code. We had created the AHRS twice.

closed 2018-02-06 by vanjac

virtuald, 2018-02-06:
Heh, that's an interesting effect.