robotpy/robotpy-wpilib-utilities #130: magicbot: Running teleop code during autonomous

virtuald, 2019-02-17:
We should have an official prescribed way to do this. Ideas?

auscompgeek, 2019-02-17:
Current (terrifying) workaround:

```python
class Robot(magicbot.MagicRobot):
  def createObjects(self):
    self.robot = self
```

```python
class TeleopSandstorm:
  MODE_NAME = "Teleop"

  robot: magicbot.MagicRobot

  def on_enable(self): pass
  def on_disable(self): pass

  def on_iteration(self, _time_elapsed):
    self.robot.teleopPeriodic()
```

ArchdukeTim, 2019-02-17:
What if you want to run an autonomous, but interrupt the state machine or whatever when someone takes over the controls?

virtuald, 2019-02-17:
It seems that the autonomous should just be an automation that could be triggered from teleop in that case?

PR #132 (2019-02-19): Add directive to allow teleop code to be executed during autonomous
* Fixes #130
closed 2019-02-19 by virtuald