robotpy/robotpy-wpilib-utilities #142: AttributeError: "SimpleWatchdog" object has no attribute "printEpochs"

CrispyBacon1999, 2019-03-21:
A CSA needed help at an event where the team's code is crashing in autonomous. They aren't using a selector for it and are just running teleop code in autonomous and are getting errors on code pushed in commit #141 on line 273. I suggested to revert back to the previous version of robotpy just to be sure that it's just that change that broke it.

Console from driver station is included. They updated to .3 after their last competition that was running .2

[image]

virtuald, 2019-03-21:
Updating to robotpy-wpilib-utilities 2019.0.9 addresses this issue.

closed 2019-03-21 by virtuald

JDuskey, 2019-03-21:
How do we upgrade in the future to avoid problems like this? I don't know what packages I need to update every time a new release is given. pip install --upgrade pyfrc does not upgrade everything.

auscompgeek, 2019-03-21:
`robotpy-installer download-robotpy` will grab the core RobotPy packages, including this, for the roboRIO.

virtuald, 2019-03-22:
Hm, it might not, since it would use -U. We need to update the documentation/implementation for pip.

virtuald, 2019-03-22:
As @auscompgeek pointed out to me ... not very long ago, but I forget when... I think we need to tell people to do `pip install -U --upgrade-strategy=eager pyfrc`. 

When using robotpy-installer, it would need to do this also.

auscompgeek, 2019-03-22:
`pip download` doesn't have an upgrade option, so it'll always download the latest version. `robotpy-installer install-robotpy` lists each package as well, so they will get upgraded regardless.

We do need to update the documentation for local development though.