robotpy/robotpy-wpilib-utilities #144: New command-based framework

auscompgeek, 2019-09-29:
WPILib 2020 has a rewrite of the command framework. We should think about how the commandbased wrapper here should interact with it.

My thoughts:

- Cancel robotpy/robotpy-wpilib#195, as the new command framework seems to actually be sane.
- We deprecate the existing commandbased package as loudly as possible.
- We then create a new package `robotpy_ext.command` (name bikesheddable) with a new `CommandBasedRobot` that uses the new scheduler.
- Any of the control flow helper stuff can be ported by whoever wants to.

Thoughts?

/cc @FRC2539 @Tyler-Duckworth

Tyler-Duckworth, 2019-09-30:
Seems like a good plan. 

What do you mean when you say deprecating commandbased as 'loudly' as possible? (I'm not quite familiar with that terminology.)

After looking through the new docs, I noticed a few things that RobotPy might handle differently:
- In the [brief examples](https://docs.wpilib.org/en/latest/docs/software/commandbased/subsystems.html#creating-a-subsystem) they give, they show of creating a Subsystem, they make the distinction between inheriting from `SendableSubsystemBase` and `Subsystem`. (The same for Commands as well) `SendableSubsystemBase` does calls some methods automatically, and they say that "advanced users seeking more flexibility" should use `Subsystem`. Do you think this distinction should be made in RobotPy?
- They do have a number of ["convenience features"](https://docs.wpilib.org/en/latest/docs/software/commandbased/convenience-features.html) which Python probably already supports, but there might need to be some tweaks to the existing code if you want to get all of them in. 
- Also, they seem to be stressing dependency injection for Commands in much the same way that MagicBot uses it. From the examples they have I haven't seen how they do it completely, but it's possible RobotPy could have a method like `create_objects()` for commandbased.

If you guys would be willing, I'd love to help in any way I can with porting over 2020's code.

virtuald, 2019-09-30:
We are always willing for people to help out, we're just not always particularly good at assigning tasks. :)

I think an open question for me (given https://github.com/robotpy/robotpy-wpilib/issues/605) is whether the new command based stuff should be pure python or compiled wrappers -- or perhaps it should be compiled wrappers with pure python overrides?

Tyler-Duckworth, 2019-09-30:
I haven't had much experience with binding in Python, so do you of any resources I could look at to learn about any of this?

virtuald, 2019-10-01:
@Tyler-Duckworth what specifically are you looking to learn? Maybe it would be useful to drop by our gitter channel and we can discuss there.

restouffer, 2019-10-17:
I am one of the programming mentors of @FRC2539, and I am willing to lend a hand in whatever way I can to get the new-style command-based programming in robotpy usable for the 2020 season. I am a fan of making command based (and everything else) as "wrappers-only" as possible, but that's in an ideal universe, so I'd say for the moment, let's try that and see if it turns out to be reasonable.

porgull, 2020-01-11:
any ETA on when 2020 rewrite will be ready? willing to help out in any way I can.

virtuald, 2020-01-11:
It's really close. Needs testing and integration. Stop by https://gitter.im/robotpy/robotpy-wpilib if you want to help.

TheTripleV, 2020-01-11:
I'd be ok with making command stuff pure-python imo. If teams want to extend those libraries with more classes, looking at a Python codebase would be helpful as a starting point.

virtuald, 2020-01-11:
The problem is that it's a lot of code, and historically we've had low off-season participation. Wrappers are a lot less work in the long run.

auscompgeek, 2021-05-16:
Update on the current state of affairs:

- `commandbased` has been moved to the [robotpy-commands-v1](https://github.com/robotpy/robotpy-commands-v1) package
  - `commandbased.flowcontrol` might be broken: robotpy/robotpy-commands-v1#14
- A small fork of the new command framework lives in [robotpy-commands-v2](https://github.com/robotpy/robotpy-commands-v2)
  - There's a small `commands2.TimedCommandRobot` wrapper that defines `robotPeriodic` to run the scheduler for you, not much else

If anyone would like to port any of the helper stuff to the commands-v2, that effort would be welcome (with tests, of course).

I'll leave this open here a bit longer for visibility.

virtuald, 2022-01-16:
OBE

closed 2022-01-16 by virtuald